Jack477/CommanderPi #2: Error message on Raspbian

Mikez333, 2020-05-31:
I installed the program on raspbian, and when I ran the program or ran test.desktop, I got the error message : Invalid desktop entry file: '/home/pi/Desktop/test.desktop'

Jack477, 2020-05-31:
Do you install it with sudo?

Jack477, 2020-06-01:
Download new version and try

Mikez333, 2020-06-01:
still says Invalid desktop entry file: '/home/pi/Desktop/test.desktop'. I am running RPI3, but when i use commander pi on twister os it worked fine (aside from the overclock making a blank screen during boot)

Jack477, 2020-06-02:
Show me your path to CommanderPi and try run by sudo ./start.sh in src folder

Mikez333, 2020-06-02:
i got python3: can't open file '/home/pi/CommanderPi/src/main.py': [Errno 2] No such file or directory
and the path is /home/pi/Desktop/CommanderPi

Mikez333, 2020-06-02:
I think I fixed it! I watched ETA prime's tutorial and I didn't move the commander pi folder to my pi user folder. That should be changed sometime in the future. thanks for your help!

closed 2020-06-02 by Mikez333