Jack477/CommanderPi #5: ENHANCEMENT REQUEST: A few suggestions

dikodahan, 2020-06-02:
First off, thank you for this! Great little app!!!

I just installed on an RPI3B+ with Raspberry Pi OS and it works without issues.
A few things I noticed though that are worth fixing:

1. I would suggest renaming `test.desktop` to `commander-pi.desktop`. This will just make it simpler and prevent any potential future conflict because of using the name test.
2. The desktop file is has a link for the icon as `logo.png`. The file does not exist. I manually updated it to icon.png and it works without any issues. A quick update to the c_desktop.py file to point to the right file will solve that.
3. After the creation of the file desktop file in `/usr/share/applications/` there is no need to leave the file on the desktop. It can be deleted in the script to keep things clean.

With that, I would also suggest a few enhancements, if I may:

- You can easily identify which RPi board and revision this is being installed on with the following command: `cat /proc/device-tree/model`.
     - In my specific example it is: `Raspberry Pi 3 Model B Plus Rev 1.3`
     - By doing that you can present the relevant recommended overclock options compatible with each board type. There are plenty of online references to recommended overclocking per board type.
     - You could also present the recommended and extreme options in text based on board type.
- Most tech savvy people will know how to edit the `config.txt` file to make this happen. I would assume this app would greatly assist the less techie people. With that, I would suggest having a few simple one click button for all 3 parameters. Like; **Basic**, **Recommended**, **Extreme**.
     - Clicking on one of these will set all parameters in a single click with a warning prompt.
     - I would still leave the manual entry options for the more advanced users. It is awesome to have it!
- Having the app present the currently set overclocking values would be very beneficial.
- Potential future enhancements: have a few additional much used capabilities, like;
     - Change boot configuration (boot from SD, USB, PXE).
     - Update OS with a single click instead of CLI.
     - Updating Swap file size as today's SD card sizes are usually enough to include a larger size.
     - If on Pi4 with 4MB/8NB ram, create a RAM drive to increase performance as well.

Again, really great small app!!!
Just throwing it out there for a few potential enhancements to help others :)

Jack477, 2020-06-02:
**Thanks for your request.**</br> I will fix this small things which I missed.</br>

1. Idea with few simple buttons to overclock is cool, you can expect something like this in near update
2. I will add annotation about RPI3B+ support
3. I will add board model on main window
4. I was thinking about boot configuration, maybe in future who knows </br>
_Honestly I'm new in RPI, I made this app from zero so support like Yours it's very helpfull_

closed 2020-06-02 by Jack477

dikodahan, 2020-06-02:
Awesome!
One more comment on this one, since this is going to have frequent updates and additions, I suggest you add a check in code against a version update in the GitHub repo to notify user of a pending update and a click to update (runs a script to close the app, downloads & updates files, relaunch app).