Jack477/CommanderPi #6: Reccomended values WAY TOO HIGH

Mikez333, 2020-06-02:
Tried recommended values and my pi turned into 
[image]

Raymond-Ursher, 2020-06-03:
The recommended values are NOT too high...you must have a different issue because I've been over-clocked at 2000mhz on CPU and 600mhz on GPU for a month straight... it's stable.

Raymond-Ursher, 2020-06-03:
Do you have an aftermarket CPU cooler?

dikodahan, 2020-06-03:
I agree. Recommended values are just fine.
You need quality active cooling for the high values to work though.

Mikez333, 2020-06-03:
I have a fan and heatsink. I am using an RPI 3 though.

dikodahan, 2020-06-03:
That's your issue then!
the listed recommended numbers are for RPi4. look up online for the recommended numbers for your Pi model.
@Jack477 has an enhancement request to recognize your Pi model and add recommended ones accordingly.

Mikez333, 2020-06-03:
https://pimylifeup.com/raspberry-pi-overclock/ for overclock info for pi versions excluding 4

Jack477, 2020-06-03:
This app is designed for Raspberry Pi **4** ! I will add in future Pi 3 support but now all of thing are made for Pi 4! It could works on Pi 3 sure, but it's made for Pi 4. I will add auto detect which Raspberry user have and then it will be showing proposed values for each Pi model.

closed 2020-06-04 by Jack477