Jack477/CommanderPi #7: Commander Pi won't start when using xrdp to remote in to Pi

nikescar1, 2020-06-04:
```
pi@raspberrypi:~/CommanderPi/src $ sudo python3 /home/pi/CommanderPi/src/main.py > a.txt
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 22, in <module>
    main()
  File "/home/pi/CommanderPi/src/main.py", line 18, in main
    start = g.Window()
  File "/home/pi/CommanderPi/src/gui.py", line 193, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":10.0"
```
When remoting in and trying to run the program I get the above error. I solved it using [THIS](https://www.raspberrypi.org/forums/search.php?author_id=8878&sr=posts) post.

Run `sudo cp ~pi/.Xauthority ~root/` in the terminal. Then CommanderPi starts as expected.

Jack477, 2020-06-04:
Hmm, try with new version of CommanderPi

nikescar1, 2020-06-05:
It's not a problem with your program, it has to do with tkinter not finding the display because of some sort of X11 display permissions.

Jack477, 2020-06-05:
Ok so I close issue

closed 2020-06-05 by Jack477

ChaosBlades, 2020-06-07:
Seems you have to run that command on every reboot. Very annoying.