Jack477/CommanderPi #8: Unable to start program

Narbife, 2020-06-12:
Hello!
I'm unable to start your program after installing it. If I try to run it from the desktop nothing happens, and if I try to run it from the terminal sudo ./start.sh, the following error message appears: "ModuleNotFoundError: no module named 'psutil'.

Thank you in advance for your time. 
[image]

Narbife, 2020-06-12:
Here's a picture of the error.
[image]

Jack477, 2020-06-12:
Try this command `sudo pip install --upgrade psutil`

prezftb09, 2020-06-12:
I am having the same issue and I tried the above command to no avail.

Narbife, 2020-06-12:
I tried that as well, but it unfortunately did not fix my issue. What eventually did was following the video bellow. After fully updating everything with the commands he showed and rebooting the system, I was able to use the software. 

https://youtu.be/RBpAkTvBbYg

The last two probably fixed the issue.

closed 2020-06-12 by Narbife