Jack477/CommanderPi #9: Overclock not applied

Mikez333, 2020-06-12:
I tried using the overclock option, and then checked the bottom of config.txt. It added my overvoltage and gpu_freq, but the arm_freq was nowhere to be seen! I ran `vcgencmd` measure_clock arm`, and the frequency was 1200.126 MHz. Maybe you forgot some code? I am running turbo mode, which I believe won't affect the frequency.

Jack477, 2020-06-13:
It's not on bottom in config.txt but in center. Do you use Pi 4?

Mikez333, 2020-06-13:
Nope. Here is config.txt file:
`# For more options and information see
# http://www.raspberrypi.org/documentation/configuration/config-txt.md
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
[all]
dtoverlay=vc4-fkms-v3d
#Overclock-Commander Pi
sdram_freq=500
force_turbo=1
gpu_freq=500
over_voltage=6
[all]
#dtoverlay=vc4-kms-v3d
`

Jack477, 2020-06-13:
**CommanderPi is desgined for Pi 4**, sure it should works on Pi3 but all stuff what I did it's for Pi 4.</br> In future I will work about compatibility with older Pi. </br>
`I am running turbo mode` - It's unnecessary, but it's strange bc CommanderPi should add arm_freq to config.txt. Turn off turbo mode and try again.

Mikez333, 2020-06-13:
still wont work

Mikez333, 2020-06-13:
if you want, I can email you vnc credentials so u can test code on rpi3.

Mikez333, 2020-06-13:
Oh wait I fixed it I had an edited config.txt so I got a fresh config.txt off the RPI forums and it worked

closed 2020-06-13 by Mikez333

Jack477, 2020-06-13:
Ohh i know why. Bc RPI4 default has a line with `armfreq=` in config.txt, idk how abut Pi3 but problem was there when CommanderPi **edited** line, only `gpu_freq` and voltage was written as new line but `arm_freq` is always default in it so CommanderPi only can edit this line

Jack477, 2020-06-13:
On soon update I will add a little fix for that issue :-)

Mikez333, 2020-06-13:
Also when I click the update button, it just closes Commander Pi. Is that normal?

Jack477, 2020-06-13:
Yes because it is download a new version from github