Jack477/CommanderPi #12: The program does not start

aleksrovinski, 2020-07-08:
I encountered a problem after updating the program, it stopped starting with the following error:
Traceback (most recent call last):
   File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
     import gui as g
   File "/home/pi/CommanderPi/src/gui.py", line 8, in <module>
     import theme as th
   File "/usr/local/lib/python3.7/dist-packages/theme.py", line 3
     print ('This is suroegin's package - theme')
                             ^
SyntaxError: invalid syntax

Jack477, 2020-07-08:
You should reinstall manually, just download all files from /src on github and put them to your Commander/src folder because updating from app menu isn't working at now

aleksrovinski, 2020-07-08:
> Вам следует переустановить вручную, просто загрузите все файлы из / src на github и поместите их в папку Commander / src, потому что обновление из меню приложения сейчас не работает

thank, this is helped

closed 2020-07-08 by aleksrovinski