Jack477/CommanderPi #13: cant run on lubuntu

Pradhyu-18, 2020-07-12:
tried to install on lubuntu but failed

Jack477, 2020-07-12:
Show me error, you can't run ./install.sh or you can't run app?

Blak-DK, 2020-07-13:
in ubuntu we get this error:
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Desktop/commanderpi.desktop'
chmod: cannot access '/home/pi/Desktop/commanderpi.desktop': No such file or directory

Jack477, 2020-07-14:
My bad, I forgot to set all paths as relative. Fixed. You should remove all old files and install new.
Tell me if it works fine
https://drive.google.com/file/d/1x1ZTkqMQpGkp8CQa2MY4aljvmVIJFaRd/view?usp=sharing

Pradhyu-18, 2020-07-15:
still not working

Jack477, 2020-07-15:
What kind of error do you have?

Pradhyu-18, 2020-07-15:
root/ path error..

Jack477, 2020-07-15:
Do you use sudo to install? Don't use it. Type ./install.sh without sudo

Pradhyu-18, 2020-07-15:
above method worked and created a desktop shortcut.........but it's not working (not opening the app)

Jack477, 2020-07-15:
Type this and show me error `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`

Pradhyu-18, 2020-07-15:
Exist and read
28.9
Total: 28.95
Used: 8.37
Free: 20.56
Traceback (most recent call last):
  File "/home/pradhyu/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pradhyu/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/pradhyu/CommanderPi/src/resources.py", line 180, in <module>
    f = open(config_path)
FileNotFoundError: [Errno 2] No such file or directory: '/boot/config.txt'

Jack477, 2020-07-15:
Are you using RaspberryPi 4 or 3? This app is designed for PI4 and it should works on 3b, it's not for desktop or others sbc.

Pradhyu-18, 2020-07-15:
i am using raspberry pi 4

Jack477, 2020-07-15:
Where is config.txt in lubuntu?

Pradhyu-18, 2020-07-15:
in system-boot

Jack477, 2020-07-15:
Give me a path and show file content

Pradhyu-18, 2020-07-15:
[image]

Pradhyu-18, 2020-07-15:
[image]

in ubuntu instead of setting overclock parameter in config.txt  user sets this parameter in usercfg.txt

Pradhyu-18, 2020-07-15:
https://www.youtube.com/watch?v=SzofbSAU1j8

as described in this video on 5:00

Blak-DK, 2020-07-15:
Fresh install  of ubuntu-mate-20.04-beta1-desktop-arm64+raspi 
Installing picommander with 
1.    chmod +x install.sh
2.    ./install.sh

error
  File "c_desktop.py", line 11, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/home/brian/Desktop/commanderpi.desktop'
chmod: cannot access '/home/brian/Desktop/commanderpi.desktop': No such file or directory

No desktop shortcut.

Pradhyu-18, 2020-07-16:
> Fresh install of ubuntu-mate-20.04-beta1-desktop-arm64+raspi
> Installing picommander with
> 
> 1. chmod +x install.sh
> 2. ./install.sh
> 
> error
> File "c_desktop.py", line 11, in
> f = open(d_dir, "w+")
> FileNotFoundError: [Errno 2] No such file or directory: '/home/brian/Desktop/commanderpi.desktop'
> chmod: cannot access '/home/brian/Desktop/commanderpi.desktop': No such file or directory
> 
> No desktop shortcut.

Extract commanderpi in /home/brian in stead of desktop....
And run Sudo chmod +x install.sh in commanderpi folder
Also ./install.sh without sudo

Jack477, 2020-07-16:
It's strange, try this: https://drive.google.com/file/d/1qIAX5WBxxerupzHz1RtS-IAjJZ3Qofn6/view?usp=sharing
I will install ubuntu on Pi today and I will do some changes for compatibility

Pradhyu-18, 2020-07-16:
hey it worked by typing this command you gave sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}.......
but not run by the desktop app shortcut that install.sh creates

Jack477, 2020-07-16:
Hmm delete app shortcut manually and try again `./install .sh` </br>Desktop shortcut should run start.sh file which use `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`. </br>If it doesn't work too, try go to CommanderPi/src and type `./start.sh`

Pradhyu-18, 2020-07-17:
i edited the commanderpi.desktop shortcut......
edited Terminal=true....every time you need to enter password to run it...
chmod not working because of ./install.sh has no privileges

Jack477, 2020-07-17:
Oh it's matter of ubuntu, beacuse you need to enter password to run sudo, in raspbian you don't
https://phpraxis.wordpress.com/2016/09/27/enable-sudo-without-password-in-ubuntudebian/

Pradhyu-18, 2020-07-19:
now working perfectly......can you change the bootloader path for ubuntu?

Jack477, 2020-07-20:
I will install ubuntu on my pi and try to make it compatible

wjgeorge, 2020-08-31:
this will certainly not run on Ubuntu as configured. It depends on vcgencmd, not part of the Ubuntu distribution.

the code doesnt check if the command isnt found:

 CommanderPi/src/start.sh 
Traceback (most recent call last):
  File "/home/ubuntu/CommanderPi/src/main.py", line 22, in <module>
    main()
  File "/home/ubuntu/CommanderPi/src/main.py", line 18, in main
    start = g.Window()
  File "/home/ubuntu/CommanderPi/src/gui.py", line 671, in __init__
    refresh()
  File "/home/ubuntu/CommanderPi/src/gui.py", line 661, in refresh
    ptext = rs.refusage()
  File "/home/ubuntu/CommanderPi/src/resources.py", line 130, in refusage
    buff=int(buff)
ValueError: invalid literal for int() with base 10: 'gen'

Jack477, 2020-10-18:
Added path for Ubuntu config.txt

paulwratt, 2020-10-24:
@wjgeorge `vcgencmd` should still be included, because its the Raspberry Pi version (just like they have to include the other RPi specific libraries normally found in `/opt/vc`)

paulwratt, 2020-11-03:
`vcgencmd` _is_ included with Ubuntu 20.10 on RPi4 (I saw someone type the command on a YT review)

Jack477, 2020-11-03:
Oh so it is included in 20.10<=version yes? Could you test is it works then?

closed 2021-05-28 by Jack477