Jack477/CommanderPi #18: Cannot run ./install.sh

Ashutoshkvdlw123, 2020-08-29:
I sucessfully installed commanderPi the first time but I was unable to run so I had to reinstall, I accidently deleted ```commanderpi.desktop``` from ```/usr/share/applications```. After that I moved CommanderPi directory from``` /Downloads``` to ```/home/pi```, now running ```sudo ./install.sh``` returns this error message:

```
Save desktop shortcut to /root/Desktop/commanderpi.desktop
Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/root/Desktop/commanderpi.desktop'
chmod: cannot access '/root/Desktop/commanderpi.desktop': No such file or directory
```
How do I fix it?

Surfxman, 2020-08-29:
Rather than raise basically the same issue, i thought i would add. I hope that is ok.

First great product!
Have installed this fine on raspbian and then rpi os 32 bit.
on the beta version of rpi os 64 bit, when i go to install, then i got multiple errors.
Your current master fixed a lot of the issues
hence why i have one last issue which is as below and the same as ashutoshkvdlw123
Thank you

Save desktop shortcut to /root/Desktop/commanderpi.desktop
Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/root/Desktop/commanderpi.desktop'
chmod: cannot access '/root/Desktop/commanderpi.desktop': No such file or directory

Ashutoshkvdlw123, 2020-08-29:
> Save desktop shortcut to /root/Desktop/commanderpi.desktop

I don't know how to do that. That's the problem.

Surfxman, 2020-08-29:
Hi my comments were geared so Jack477 would see there is an issue with more than one person and would hopefully guide us in the right direction

Jack477, 2020-08-29:
Hi, you should install it without sudo beacuse then it create shortcuts for root user. Just use `./install.sh` . Tell me if it works. </br> There is my fault beacuse in `readme.txt` I tell to use '`sudo ./install.sh`' but I removed it yesterday.

Surfxman, 2020-08-29:
No worries, will try that. Thank you for coming back so quickly

Surfxman, 2020-08-29:
Hi Jack477, did as stated. script pushes through on terminal and a desktop image appears now. However when i double click on the icon and choose either execute or execute from terminal i cant get the GUI to pop up

Jack477, 2020-08-29:
Maybe it's old desktop shortcut created when you use sudo to install, remove it and use ./install.sh it should create new one. </br>
What OS do you have?

Surfxman, 2020-08-29:
i am using rpi os 64 bit
sorry how do i remove it - just delete

Jack477, 2020-08-29:
Yes, just delete

Surfxman, 2020-08-29:
so i deleted everything, i re-downloaded your master followed the script including revision and still no GUI?

Jack477, 2020-08-29:
Try to run by terminal `sudo python3 /home/pi/CommanderPi/src/main.py /home/pi` is it works?

Surfxman, 2020-08-29:
pi@raspberrypi:~ $ sudo python3 /home/pi/CommanderPi/src/main.py /home/pi
python3: can't open file '/home/pi/CommanderPi/src/main.py': [Errno 2] No such file or directory
pi@raspberrypi:~ $

Jack477, 2020-08-29:
I know, rename `CommanderPi-master` to `CommanderPi`

Surfxman, 2020-08-29:
just bear with me, i was trying the same thing but the other way

Surfxman, 2020-08-29:
so after renaming i get
Invalid desktop entry file: '/home/pi/Desktop/commanderpi.desktop'

Surfxman, 2020-08-29:
ok, big sigh of relief. i started from scratch. i extracted the master. renamed to CommanderPi then did the install script and hey presto

Surfxman, 2020-08-29:
and your script from the terminal also brought up the GUI

Jack477, 2020-08-29:
I'm glad to hear it, on github I have section 'Download' where is my own link, I don't prefer to download master, but I do some change for github-master download to rename this folder automaticly.

closed 2020-08-29 by Jack477