Jack477/CommanderPi #20: Icon request

Botspot, 2020-09-21:
I am developing a RPi app store and I plan to include this software.
When users click "install", the app store will then clone your repository and run the recommended install commands.

Only thing missing is the icon: I need a 24x24 icon and a 64x64 icon.
Now I could crop your icon into a square and scale it, but that doesn't look that crisp after scaling. So if you could provide me with some native-size icons, that would be great.

Jack477, 2020-09-22:
Sure, I will send message to grayduck to do that, when he done I send you and I want to be sure that app will be for free in your store.

Jack477, 2020-09-22:
https://drive.google.com/file/d/15XP7CM3Ybg3hEB_pt8yIvdftZzuKgdFD/view?usp=sharing

Botspot, 2020-09-22:
Oh, yes, everything in the app store will be free.

Botspot, 2020-09-22:
Thanks for those icons. I think we're good to go!

Could you do me a favor? Could you contact Grayduck and ask him if he'd consider adding [Pi-Power-Tools](https://github.com/Botspot/Pi-Power-Tools) to TwisterOS

Botspot, 2020-11-04:
Would you consider adding this little badge to your README?
```
[![badge](https://github.com/Botspot/pi-apps/blob/master/icons/badge.png?raw=true)](https://github.com/Botspot/pi-apps)
```
[image]

closed 2021-03-27 by Jack477