Jack477/CommanderPi #23: No desktop

bingy0744, 2020-09-26:
no fiile  commanderpi.desktop

Jack477, 2020-09-26:
What do you mean? Have you delete it?

bingy0744, 2020-09-26:
Install screen:-

pi@raspberrypi:~ $ cd CommanderPi
pi@raspberrypi:~/CommanderPi $ sudo ./install.sh
/home/pi/CommanderPi
Checking for tkinter: Status: install ok installed
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pil is already the newest version (5.4.1-2+deb10u2).
python3-pil.imagetk is already the newest version (5.4.1-2+deb10u2).
The following package was automatically installed and is no longer required:
  rpi-eeprom-images
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
/home/pi/CommanderPi
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/pi/CommanderPi/src/start.sh
Icon=/home/pi/CommanderPi/src/icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /root/Desktop/commanderpi.desktop
Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/root/Desktop/commanderpi.desktop'
chmod: cannot access '/root/Desktop/commanderpi.desktop': No such file or directory

Does this help?

bingy0744, 2020-09-26:
Sorry closed it by mistake. My first using Github issues.

Jack477, 2020-09-26:
Install it without sudo, just `./install.sh`

paulwratt, 2020-10-24:
that was in the "other" readme, has it not been changed yet?

This is what I was talking about when I said "failing gracefully" - atm it does not. It just means to catch corner cases programatically, and _not_ stop from running properly (ie dont crash)

closed 2021-05-29 by Jack477