Jack477/CommanderPi #27: Latest version on berryboot causing error

markybill, 2020-10-07:
Upgraded to the latest version now get the following trace back running from terminal
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pi/CommanderPi/src/gui.py", line 5, in <module>
    import bootloader as btl
  File "/home/pi/CommanderPi/src/bootloader.py", line 21, in <module>
    ad = get_actual_version()
  File "/home/pi/CommanderPi/src/bootloader.py", line 12, in get_actual_version
    month = MONTHS[x_version_split[0]]
KeyError: 'unknown'

It would appear that vcgencmd bootloader_version is returning unknown, changed to vcgencmd version and all works ok.

I am using latest raspberry os but booting from berryboot.

Jack477, 2020-10-07:
What do you see after type `vcgencmd bootloader_version` in terminal?

markybill, 2020-10-07:
This is what I get for the two commands

pi@raspberrypi:~ $ vcgencmd bootloader_version
unknown
pi@raspberrypi:~ $ vcgencmd version
Aug 19 2020 17:41:31 
Copyright (c) 2012 Broadcom
version e90cba19a98a0d1f2ef086b9cafcbca00778f094 (clean) (release) (start_x)
pi@raspberrypi:~ $

Hope that helps

Jack477, 2020-10-07:
Hmm it's strange, I depends all on official doc https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md </br>
`vcgencmd version` "Displays the build date and version of the firmware on the VideoCore." so it's not bootloader version?
(source https://www.raspberrypi.org/documentation/raspbian/applications/vcgencmd.md)

I think you should try update eeprom</br>
`sudo apt update`</br>
`sudo apt full-upgrade`</br>
`sudo reboot`

markybill, 2020-10-07:
I read the documentation for the booteeprom and realized I forgot to mention that I am running the Commander on a Pi 3 model B+ and not a Pi 4, I guess this could be the issue. It works fine otherwise on the Pi3 and I find the info displayed useful.

Jack477, 2020-10-07:
Yes it could be issue. After 1.0 release I will do some work for compatibility with PI 3. Now all is designed for PI 4.

KinoZampie, 2020-10-16:
Oops! Looks like this error might be my mistake.
As it stands, looks like it's only the Pi 4 that has an EEPROM bootloader.
(https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md)

I can confirm on my Raspberry Pi 2 on an installation of Raspberry Pi OS I also get `unknown` in response to `vcgencmd bootloader_version`

Maybe it would be a good idea to add a condition to check for a bootloader upon startup and if it is a Pi 1-3 (aka bootloader_version is unknown) just hide the bootloader section entirely in the gui?

Of course, this can all be added in the 1.0 release, as you said.

closed 2020-10-24 by Jack477