Jack477/CommanderPi #33: won't start

Itai-Nelken, 2020-10-26:
commanderpi won't start from menu or desktop shortcut, so I reinstalled, and then the desktop shortcut was owned by 'root' so I opened the file manager as root and changed it, but it still won't run.
I tried running start.sh from the terminal and I got some errors:

Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pi/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/pi/CommanderPi/src/resources.py", line 188, in <module>
    f = open(config_path)
FileNotFoundError: [Errno 2] No such file or directory: ''

Jack477, 2020-10-26:
How you install it? With just `./install.sh`?

daemonsneverstop, 2020-10-27:
Same here, I used Eta Prime's tutorial https://www.youtube.com/watch?v=97ZaWYjsZdQ, HOWEVER I am running Kali on my Pi 4 so maybe this doesn't work for Kali pi 4 or maybe there's a certain way I need to open it up that I don't know about. I am pretty new to Linux so I'm sure it is maybe something I am not doing right.

Jack477, 2020-10-27:
Hmm so you are on Kali, type this in terminal `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`</br>
Then tell me what do you see.
`

Itai-Nelken, 2020-10-29:
@Jack477 
> How you install it? With just `./install.sh`?

Yes.

Jack477, 2020-10-30:
This error says that it can't find `config.txt` file, do you have `/boot/config.txt` ?

daemonsneverstop, 2020-10-31:
> Hmm so you are on Kali, type this in terminal `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`
> 
> Then tell me what do you see.
> `

Exist and read
30.9
Total: 29.16
Used: 8.63
Free: 19.26
/boot/config.txt
/bi
/s
: 1:
/bi
/s
: 1:
/bin/sh: 1:
/root/CommanderPi/src
File is exist!
Traceback (most recent call last):
  File "/root/CommanderPi/src/main.py", line 22, in <module>
    main()
  File "/root/CommanderPi/src/main.py", line 18, in main
    start = g.Window()
  File "/root/CommanderPi/src/gui.py", line 579, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.8/tkinter/__init__.py", line 2261, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: no display name and no $DISPLAY environment variable

Jack477, 2020-10-31:
@daemonsneverstop you have installed CommanderPI on `/root` , you should do it at `/home/pi`. Don't use it as root.

daemonsneverstop, 2020-10-31:
> @daemonsneverstop you have installed CommanderPI on `/root` , you should do it at `/home/pi`. Don't use it as root.

Okay so I deleted everything and tried it again as kali not root, I downloaded the zip file and it is saved to /home/kali/CommanderPi-master/ , I run the commands to make it executable and install and everything seemed fine, the desktop icon appears with a lock on the side of it and when I click on it nothing happens, I ran the command you mentioned before and this is what it shows now,

"python3: can't open file '/home/kali/CommanderPi/src/main.py': [Errno 2] No such file or directory"

Thanks for you help so far though

KinoZampie, 2020-10-31:
In home/kali try renaming the folder CommanderPi-master to CommanderPi

Itai-Nelken, 2020-11-01:
I reinstalled and everything is working again. Maybe 5he error happens when using git clone, but when downloading from the Google drive link, the after renamin the directory to CommanderPi everything works.

closed 2020-11-01 by Jack477

Itai-Nelken, 2020-11-24:
won't start on a clean install of TwisterOS, even after reinstall. here is the log.
[log.txt](https://github.com/Jack477/CommanderPi/files/5588392/log.txt)

Itai-Nelken, 2020-11-24:
Won't start on a clean install of TwisterOS 1.9.1, so I reinstalled but I still get the same error. 
[log.txt](https://github.com/Jack477/CommanderPi/files/5588406/log.txt)