Jack477/CommanderPi #35: Install broken

danielktdoranie, 2020-11-09:
Install not working as it did before. 

Fresh install of Raspberry Pi OS 32-bit, updated.

Here's the install output:

pi@raspberrypi:~/CommanderPi-master $ sudo chmod +x install.sh
pi@raspberrypi:~/CommanderPi-master $ sudo ./install.sh
/home/pi/CommanderPi-master
Checking for tkinter: Status: install ok installed
./install.sh: 6: [: unexpected operator
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-pil is already the newest version (5.4.1-2+deb10u2).
python3-pil set to manually installed.
Suggested packages:
  python-pil-doc python3-pil.imagetk-dbg
The following NEW packages will be installed:
  python3-pil.imagetk
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 65.0 kB of archives.
After this operation, 92.2 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.heanet.ie/mirrors/raspbian/raspbian buster/main armhf python3-pil.imagetk armhf 5.4.1-2+deb10u2 [65.0 kB]
Fetched 65.0 kB in 1s (92.1 kB/s)            
Selecting previously unselected package python3-pil.imagetk:armhf.
(Reading database ... 95104 files and directories currently installed.)
Preparing to unpack .../python3-pil.imagetk_5.4.1-2+deb10u2_armhf.deb ...
Unpacking python3-pil.imagetk:armhf (5.4.1-2+deb10u2) ...
Setting up python3-pil.imagetk:armhf (5.4.1-2+deb10u2) ...
/home/pi/CommanderPi-master
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/pi/CommanderPi-master/src/start.sh
Icon=/home/pi/CommanderPi-master/src/icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /root/Desktop/commanderpi.desktop
Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/root/Desktop/commanderpi.desktop'
chmod: cannot access '/root/Desktop/commanderpi.desktop': No such file or directory

Same issue when I tried ./install (without sudo). Re-imaged SD Card and tried the process all over again using sudo ./install and have the same issue

Jack477, 2020-11-09:
What path have your installation? Is it /home/pi/CommanderPi? You should rename folder from CommanderPi-master to CommanderPi, then `./install.sh`

danielktdoranie, 2020-11-09:
When I extracted it, that's the name it had, maybe you changed the name of the folder before you zipped it?

Jack477, 2020-11-09:
My bad, I zipped github version which automaticly added '-master' to folder name. Is it works for you now?

danielktdoranie, 2020-11-09:
> My bad, I zipped github version which automaticly added '-master' to folder name. Is it works for you now?

Yep! All good now! Thank you 👎

Jack477, 2020-11-09:
Your welcome! :)

closed 2020-11-09 by Jack477