Jack477/CommanderPi #39: Doesn't start

Jegoni, 2020-11-28:
Hello! I installed it I can see the icons, no error code but when I try to start it, it does't start. Just loading, but nothing happens.

Jack477, 2020-11-28:
Type this in terminal sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
Tell me what you see then

Polygonboy, 2020-11-29:
I got no such file in the directory.
I am having the same problem

Jegoni, 2020-11-29:
I got:
Creating config...
Here is app-1 Version 0.7.2
Country code is:
country HU: DFS-ETSI
68.8
Total: 29.01
Used: 18.93
Free: 8.58
2.1GHz
Can't find RaspberryPi config file!

Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pi/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/pi/CommanderPi/src/resources.py", line 245, in <module>
    f = open(config_path)
FileNotFoundError: [Errno 2] No such file or directory: ''

Jack477, 2020-11-29:
What is your config.txt file path?

Jegoni, 2020-11-29:
/media/pi/boot/config.txt

Jack477, 2020-11-29:
Strange path, what OS are you using?

Jegoni, 2020-11-29:
I am using Raspberry os 32-bit but with berryboot. And I think it messed it up.

Jegoni, 2020-11-29:
> Strange path, what OS are you using?
> 
> I have added this path to CommanderPi, download new version and try
> https://drive.google.com/file/d/12Lo9WWjix-88N6P-C8aeISUYX_xOlcKg/view?usp=sharing

This doesn't work too, but no problem because from berryboot I can change the config file. But thank you.

arthertz, 2020-11-29:
I have a potential fix for this issue. If you look in your start.sh file, you should see a little '>' after the second ${HOME}.
I didn't understand why it was there, so I tried removing it, and then the program started normally.

The old start.sh looks like this:
`#!/bin/sh
sudo python3 %{HOME}/CommanderPi/src/main.py ${HOME}> /var/log/commanderpi.log`
And your new start.sh would look like this:
`#!/bin/sh
sudo python3 %{HOME}/CommanderPi/src/main.py $HOME /var/log/commanderpi.log`

Does anyone know why the '>' is there, or am I going crazy?

pasdechance, 2020-11-30:
thanks @arthertz

Jack477, 2020-11-30:
@arthertz this '>' is for create log file if program has crashed.

jaymoore962, 2020-11-30:
@Jack477 my commander pi wont open two i have the latest version and it just loads and never opens up

arthertz, 2020-12-01:
@Jack477 Is it possible there is a platform-specific reason why that would cause a problem? I am certain that removing it fixed my issue. I wonder if there is a more portable way to do the same thing.

Jack477, 2020-12-01:
I checked that, problem was `>` because this operator need sudo permissions to create file at `/var/log/`. On my Pi I was have a setup permissions for that. Removed that. Now it works. I'm sorry, use new download version and reinstall is needed.

arthertz, 2020-12-01:
Awesome! I love your program, thank you for fixing this so quickly.

midengc8, 2020-12-14:
I have the same problem, installed ok and have the CommanderPi logo on the desktop, but if you run it, it will not show the CommanderPi window. I believe there is a path issue on my setup. On this set up, I need to enter password on any sudo type commands.  Will it cost an issue  on your code? Thanks.

closed 2020-12-15 by Jack477