Jack477/CommanderPi #40: Not working

memidhun, 2020-12-14:
I am using kali linux in rasberry pi 4 model B . I downloaded commander pi unzip it in home directory and all the logos shown up but it was not starting so i thought the commanderpi folder should be in the root directory.. i don't know how to uninstall so i just deleted the folder and created a new one in root directory, and once again i did ./install.sh and i got the same problem .
[image]

also can anyone tell me how to completely uninstall this CommanderPi?

Jack477, 2020-12-14:
To uninstall just delete folder with files. Don't install it as root user.

memidhun, 2020-12-14:
> To uninstall just delete folder with files. Don't install it as root user.

Still it's not runnung..I tried clicking the desktop icon

Jack477, 2020-12-14:
Type this in terminal and tell me what you see `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`

memidhun, 2020-12-15:
> sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}

Traceback (most recent call last):
  File "/home/kali/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/kali/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/kali/CommanderPi/src/resources.py", line 16, in <module>
    home_path = sys.argv[1]
NameError: name 'sys' is not defined

Jack477, 2020-12-15:
Try this version, I've added manually sys import
[CommanderPi.zip](https://github.com/Jack477/CommanderPi/files/5695614/CommanderPi.zip)

memidhun, 2020-12-15:
> Try this version, I've added manually sys import
> [CommanderPi.zip](https://github.com/Jack477/CommanderPi/files/5695614/CommanderPi.zip)

Traceback (most recent call last):
  File "/home/kali/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/kali/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/kali/CommanderPi/src/resources.py", line 188, in <module>
    print(refusage())
  File "/home/kali/CommanderPi/src/resources.py", line 165, in refusage
    buff=int(buff)
ValueError: invalid literal for int() with base 10: 'gen'

Jack477, 2020-12-15:
Try this, if it wont work type this in terminal and show me your output `vcgencmd measure_clock v3d` I guess problem depends on Kali beacuse CommanderPi is designed for Debian/Raspbian.
[CommanderPi.zip](https://github.com/Jack477/CommanderPi/files/5697723/CommanderPi.zip)

closed 2021-05-28 by Jack477