Jack477/CommanderPi #41: commanaderPi won'

midengc8, 2020-12-14:


Jack477, 2020-12-15:
Are you using Raspberry Pi 4? What is your path to CommanderPi and did you installed it as user not root?

midengc8, 2020-12-15:
yes, using a Raspberry Pi 4.   Path is Home/Pi/CommanderPi.  I have Octoprint running also.

Jack477, 2020-12-15:
Type this in terminal and tell me what you see sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}

midengc8, 2020-12-15:
pi@octopi:~/CommanderPi $ sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
[sudo] password for pi: 
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pi/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/pi/CommanderPi/src/resources.py", line 5, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'

Jack477, 2020-12-16:
Install this module with `pip install psutil`

midengc8, 2020-12-16:
Installed, but looks like still missing, CommanderPi still not working, same as before.    Please look at the following :  Need in a different path?   Thanks for the help.

pi@octopi:~/CommanderPi $ pip install psutil
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: psutil in /usr/local/lib/python2.7/dist-packages (5.7.3)
pi@octopi:~/CommanderPi $ sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 9, in <module>
    import gui as g
  File "/home/pi/CommanderPi/src/gui.py", line 4, in <module>
    import resources as rs
  File "/home/pi/CommanderPi/src/resources.py", line 5, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'

KinoZampie, 2020-12-16:
Perhaps "pip3 install psutil"?

midengc8, 2020-12-16:
Still the same, no psutil module. not sure what is going on.

midengc8, 2020-12-17:
I tried to import psutil directly, this is what I got:
pi@octopi:~/.local/lib/python3.7/site-packages/psutil $ import psutil

import-im6.q16: attempt to perform an operation not allowed by the security policy `PS' @ error/constitute.c/IsCoderAuthorized/408.

Any idea what is going on here? I am not too well verse on Linux.

Jack477, 2020-12-18:
What OS are you using?

midengc8, 2020-12-18:
Raspberry Pi OS  with desktop. I also installed octoprint.

Whiskaya, 2020-12-24:
I struggled with this for like half an hour until I *read the readme.txt file*.
Did you run install.sh *AFTER* you moved the CommanderPi folder from Downloads to your Home folder?
At least this was my mistake, and it worked right away.

midengc8, 2020-12-25:
Did all of those, same thing. The interesting part is that when I do not have Octoprint in the built, CommanderPi works. I wonder what is causing the problem with Octoprint.

Chulamin, 2021-03-09:
Hi,
It doesn't seem to like Sparky Linux. Maybe it shouldn't but most other Pi apps do....so far ;-)

gprime31, 2021-03-22:
Ya, it's still needs Sudo to run, extract it in /home/pi/
Then:
./install.sh (non Sudo) 
then cd /home/pi/ComamanderPi/src.
Then run sudo ./start.sh from terminal, if you don't run sudo it will ask anyway, works fine, just don't use desktop shortcut.

vieirae, 2022-03-19:
Here is what I get:
pi@raspberrypi:~ $ sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
Here is home_path /home/pi
Here is app-1 Version 1.0
Exist and read
Country code is:
country DE: DFS-ETSI
Here is ad: pieeprom-2021-12-02.bin
Apr 29 2021
/home/pi/CommanderPi/src
File is exist!
path is /home/pi/CommanderPi/src
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 26, in <module>
    main()
  File "/home/pi/CommanderPi/src/main.py", line 22, in main
    start = g.Window()
  File "/home/pi/CommanderPi/src/gui.py", line 673, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":10.0"

The two last lines seem to indicate the app can't access the display: but when I double click on the CommanderPi icon
I get the "Execute File" menu. But the buttons don't work.

skarai, 2022-07-20:
Here is my output - as suspected the culprit is the ongoing rdp session which holds display 10 already.
When executed directly on the desktop it works fine.
----------------------------------------------------------------------------------------------------------------------------------------
sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
Here is home_path /home/sassan
Here is app-1 Version 1.0
Exist and read
Country code is:
country 99: DFS-UNSET
Here is ad: pieeprom-2022-04-26.bin
2022/04/26 
/home/sassan/CommanderPi/src
File is exist!
path is /home/sassan/CommanderPi/src
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/sassan/CommanderPi/src/main.py", line 26, in <module>
    main()
  File "/home/sassan/CommanderPi/src/main.py", line 22, in main
    start = g.Window()
  File "/home/sassan/CommanderPi/src/gui.py", line 673, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.9/tkinter/__init__.py", line 2270, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":10.0"