Jack477/CommanderPi #42: CommanderPi installs but doesn't run on RaspberryPi OS using a RaspberryPi 4 64bit

chest069, 2020-12-23:
Hello,

I just found this program and I installed it but nothing seems to work when trying to get it to run. I have a Raspberry Pi 4 8GB 64bit system and I currently run Raspberry Pi OS. I get the following error :

pi@pi-desktop:~/Desktop $ /home/pi/CommanderPi/src/start.sh
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 22, in <module>
    main()
  File "/home/pi/CommanderPi/src/main.py", line 18, in main
    start = g.Window()
  File "/home/pi/CommanderPi/src/gui.py", line 567, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":0"

Any ideas or help in getting this running please.

Jack477, 2020-12-23:
You installed is as root, you must install it without sudo.

chest069, 2020-12-24:
> You installed it as root, you must install it without sudo.

Thanks, I restarted my Pi and then it ran no problem.

closed 2021-05-28 by Jack477