Jack477/CommanderPi #46: CommanderPi can not run for Raspberry Ubuntu20.1

xiangdongping999, 2021-01-23:
Hello sir:

I download the install zip,and installed app.
but i run this app ,this app is no reaction.
use the terminal run the start.sh not run and show some error information (pic 1)
[image]
I read some issues
used sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME} ,(pic2)
[image]
and i surety installed psutil (pic3)
[image]
How to fix ,Please help me! 
Thank you!

PS:I am from china, english is not good.

Jack477, 2021-01-23:
Hi, try with `pip3 install psutil`, also check is CommanderPi installed in your home directory as ` /home/<username>/CommanderPi ` NO AS `/root`!

xiangdongping999, 2021-01-23:
hello sir

I try with pip3 install psutil (pip3 install)
[image]
and check the file (file)
[image]
everything is ok but app not run.

thank you !

Jack477, 2021-01-23:
Hmm, try install psutil with sudo, I mean `sudo pip3 install psutil`

xiangdongping999, 2021-01-24:
hello sir:
I try install psutil with sudo , but app not run.
[image]

and i in terminal run the start.sh $ python main.py 
this time show the error information are different (resources.py)
[image]
[image]
run start,sh error line before is line6, now is line248
run main.py error line before is line6, now is line17

I need edit resoures.py ? if need edit ,how to edit ? 

Thank you!

xiangdongping999, 2021-02-03:
hello sir:

Do you have any way for fix this ?

Jack477, 2021-02-03:
Hello, what is path for your config.txt file?

Itai-Nelken, 2021-02-03:
I think its `boot/firmware/config.txt` in Ubuntu, but the file where you should change things has a different name.

Jack477, 2021-02-03:
Replace `resources.py` with this file https://drive.google.com/file/d/10D0gSCooz9w-sQ7wxkzowJM7WV-eqMWw/view?usp=sharing

xiangdongping999, 2021-02-04:
hello sir
I try this way ,but app is not run.
I download this file and changed.
I Added the config file path (pic:config)
[image]
This way is right ?
And test start.sh
[image]
And test main.py
[image]

What's wrong with me ?

Jack477, 2021-02-04:
Im sorry my bad, I sent you not realesed version of CommanderPi. This file should be fine for you https://drive.google.com/file/d/1oeypiH-ju3lQw0Y_3Z6nCoaoCZFIF6Og/view?usp=sharing

xiangdongping999, 2021-02-05:
Hello sir
Thank you very much!
run the start.sh ,software is running.
menu desktop not work
but this software can running is well

thank you~

closed 2021-02-10 by Jack477