Jack477/CommanderPi #47: Can you help, trying to install Commander Pi on RaspPi4b 8GB running Windows10, downloaded sip file but cant seem to run?

Laurenceellis, 2021-01-23:


Laurenceellis, 2021-01-23:
[image]

Laurenceellis, 2021-01-23:
anything I type after this - says it does not recognise as an operable file

KinoZampie, 2021-01-23:
Sorry but CommanderPi is designed to run on Linux operating systems, specifically Raspberry Pi OS or TwisterOS. You will not be able to run CommanderPi on Windows 10

Laurenceellis, 2021-01-23:
Arrgg OK was hoping to overclock etc. through this - will have to go back through the long route

Itai-Nelken, 2021-01-24:
> Arrgg OK was hoping to overclock etc. through this - will have to go back through the long route

@Laurenceellis use WoRCP for that, you can get it from the WoR discord server.

Jack477, 2021-01-24:
Hi like @KinoZampie said CommanderPi is designed for Linux. I don't think it would be port to WoR anytime soon

closed 2021-01-24 by Jack477