Jack477/CommanderPi #48: App opens and then crashes immediately

JRAUSMC4L, 2021-02-18:
I am new to Raspberry Pi and have a Pi4.  I downloaded the app and followed the directions.  The link is on my desktop, but when I  open  the file it will pop up the Commander Pi and then it goes away.  Suggestions?  Thank you!

Jack477, 2021-02-19:
Try to run it with following command `sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`

JRAUSMC4L, 2021-02-22:
Jack 477,

Thank you for getting back to me! I ran the command you suggested.  It spit out a long list of stuff in the terminal.  I attempted to reopen the commander pi and it was still crashing.

Jason

Itai-Nelken, 2021-02-22:
@JRAUSMC4L can you send the output of that commad? Its supposed to open CommanderPi, and @Jack477 needs the output to know whats wrong.

Jack477, 2021-02-22:
Like @Itai-Nelken said I need the terminal output of this command. @JRAUSMC4L

JRAUSMC4L, 2021-02-23:
[image]
[image]
[image]
[image]