Jack477/CommanderPi #55: CommanderPi won't boot from desktop or from Terminal on PRI4 with Raspbian

TAMUB, 2021-04-28:
When I attempt to boot from Desktop, nothing happens. 
if I run sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
I see:
Exist and read
Here is app-1 Version 0.7.2
Country code is:
country US: DFS-FCC
94.6
Total: 14.29
Used: 12.9
Free: 0.73
1.5GHz
/boot/config.txt
Here is ad: pieeprom-2021-03-18.bin
Sep  3 2020
/home/pi/CommanderPi/src
File is exist!
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 22, in <module>
    main()
  File "/home/pi/CommanderPi/src/main.py", line 18, in main
    start = g.Window()
  File "/home/pi/CommanderPi/src/gui.py", line 567, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":10.0"

Thanks for any help!

CleanMachine1, 2021-04-28:
Try without the sudo

TAMUB, 2021-04-29:
Thank you so much CleanMachine1!

closed 2021-04-29 by TAMUB

CleanMachine1, 2021-04-29:
Glad to help