Jack477/CommanderPi #56: CommanderPI doesn't have permissions to change config file

TAMUB, 2021-04-29:
I previously had an issue where I couldn't launch CommanderPi from Desktop (still can't) however I was able to get CommanderPi to launch by running: 
python3 ${HOME}/CommanderPi/src/main.py ${HOME}

However, when CommanderPi runs, it is unable to Overclock because it doesn't have permissions to modify /boot/config.txt
This is the error: PermissionError: [Errno 13] Permission denied: '/boot/config.txt'

How can I grant CommanderPi this access? I'm sure it's something really simple that I'm missing.
FYI: I'm running a RPI4 with Raspbian.

Thanks for any help! This tool is really cool!

CleanMachine1, 2021-05-06:
just try python3 ~/CommanderPi/src/main.py

also try it with a sudo before it however this might break it a little

CleanMachine1, 2021-05-06:
if you are going to try without the sudo, make sure to run sudo python3 /home/YOURUSER/CommanderPi/src/main.py

Also have a look through the code and try to find where it starts editing /boot/config.txt, you may be able to edit it and add sudo before it (whatever command it is running)

If you fail to do that then just go into a terminal and type

sudo vim /boot/config.txt
then go down to the bottom, press i to insert then
write this

over_voltage=6
arm_freq=2100
gpu_freq=700

then press [Esc] then :x - enter

(Or just use nano however why not learn a little vim :smile: )

closed 2021-06-01 by Jack477