Jack477/CommanderPi #61: Doesn’t Work on Pi 4

prashantdwivedi, 2021-06-01:
Try to open it not open

Jack477, 2021-06-01:
Could you explain what "not open"?

MagicAndre1981, 2021-06-08:
https://github.com/Jack477/CommanderPi/issues/54

Jack477, 2021-06-11:
What is path to your installed Comannder Pi, what OS do you use and do you have the newest version?

MagicAndre1981, 2021-06-11:
I'm using the normal RaspiOS on Pi4 and the app doesn't start when not in expected folder. I was also using desktop first. Now it works, this was just a hint for @prashantdwivedi .

There is also a ; missing in [install command](https://github.com/Jack477/CommanderPi#to-download-the-software)

prashantdwivedi, 2021-06-11:
Typical method installation does not work. I find it extracted on root and install then perform—that method I see on youtube.
Otherside try to install from the download folder; it did not work app.

Jack477, 2021-06-11:
@MagicAndre1981 I have added missing `;` @prashantdwivedi  now it should works fine

prashantdwivedi, 2021-06-11:
Cool, I will check it later. Thanks for the fix.

closed 2021-06-11 by Jack477