Jack477/CommanderPi #63: Cannot find the overclock option in new 1.0 version

sewichou, 2021-06-14:
I just update my Commander pi and cannot find any tools and options. How can I use overclock in this version?

Itai-Nelken, 2021-06-14:
there is a drop-down menu at the bottom, click it and all the options will expand.

sewichou, 2021-06-14:
> there is a drop-down menu at the bottom, click it and all the options will expand.
[image]
Well...still, can't find it
orz

Itai-Nelken, 2021-06-14:
weird! its there for me.

MardukNT, 2021-06-20:
Q) @sewichou, if you change themes or attempt to scroll the window down, are you then able to see the **"Advanced tools"** button?

[image]

sudo-splinter-cell, 2023-10-31:
The exact same problem is present in the latest Bookworm version of RPI OS. The below menu options are not visible. There is also no way to "extend" the windows size by pulling from the corner of the window. So this requires a bug fix please.