Jack477/CommanderPi #64: CommanderPi fails to launch after install onto TwisterOS Lite v2.0.3

MardukNT, 2021-06-20:
**System Specs:**
OS Mod: **Twister OS Lite version 2.0.3** 
OS: **Raspbian GNU/Linux 10 (buster) armv7l** 
Host: **Raspberry Pi 4 Model B Rev 1.1** 
Kernel: **5.10.17-v7l+** 
Uptime: **12 mins** 
Shell: bash **5.0.3** 
CPU: **BCM2711 (4) @ 2.000GHz** 
Memory: **776MiB / 3774MiB** 
Disk (/): **5.6G / 110G (6%)**

**Issue:**
CommanderPi v1.0 fails to launch when WiFi overlay is set to disabled in **/boot/config.txt**

**Steps to reproduce:**

- Install CommanderPi v1.0 following the Github instructions or via Pi-Apps.
- Next, in a terminal window, type in `**sudo nano /boot/config.txt**` and hit **Enter**.
- Scroll to the bottom and add `**dtoverlay=disable-wifi**`. Then **Save** and **Exit**.
- Restart system and log back in.
- Attempt to launch CommanderPi 
- Observe busy cursor briefly appears onscreen, then disappears. No GUI is ever shown.

**Troubleshooting performed (so far):**
From a Terminal session, I ran the command **`sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}`**

**Here are the results (w/emphasis added by me):**
`Here is home_path /home/pi`
**`Here is app-1 Version 1.0`**
`Exist and read`
`Traceback (most recent call last):`
`  File "/home/pi/CommanderPi/src/main.py", line 2, in <module>`
`    import gui as g`
`  File "/home/pi/CommanderPi/src/gui.py", line 4, in <module>`
`    import resources as rs`
**`  File "/home/pi/CommanderPi/src/resources.py", line 60, in <module>`**
`    ipv4wlan0 = network['wlan0'][0][1]`
**`KeyError: 'wlan0'`**

Commenting out **`dtoverlay=disable-wifi`** in /boot/config.txt and restarting, resolves the error.

**Expected behavior**
App should be able to launch and function properly, when WiFi is disabled at startup by the OS.

**Final Thoughts**
Thank you for your attention to this matter.  Kindly reach out to me, if you require further information or further testing.

--MardukNT

PJOttCanHome, 2023-02-10:
This issue appears to persist to this day.
I looked at the code in src/resources.py and at line 60 it seems to assume a 'wlan0' entry in the network[] list which would not be true when not connected to wifi. I tried to remedy it by putting the similar try/except block around that line but that didn't work as expected, either. {unlike MardukNT my Pi is simply wired to eth0, no wifi at all). This would also be an issue for completely non-connected PI's, as well, correct?

PJOttCanHome, 2023-02-10:
I got it to work by changing the "IndexError" try keywords to (IndexError, KeyError) to match the exception occurring.  This is my wlan0 code starting around line 60 in src/resources.py:

`try:
	ipv4wlan0 = network['wlan0'][0][1]
	broadcastwlan = network['wlan0'][0][3]
	macwlan0 = network['wlan0'][2][1]
except (IndexError, KeyError):
	ipv4wlan0 = None
	macwlan0 = None
	broadcastwlan = None

ipv6wlan0 = None
try:
	ipv6wlan0 = network['wlan0'][1][1]
except (IndexError, KeyError):
	ipv6wlan0 = None
`
So devs/maintainers, watchout for those assumptions that we all make. Programming challenges!?!

menotuu, 2023-09-26:
so what is your src/resources.py code now exactly?

I have the same behaviour.
I replaced "except IndexError:" with "except (IndexError,KeyError):"
but still receiving KeyError: 'wlan0'

I have deactivated wifi for power saving and using eth0 ipv4.