Jack477/CommanderPi #67: What copies the desktop file to ~/Desktop?

Botspot, 2021-08-23:
Today, us Pi-Apps developers received this error report from one of our users:

Please review, research, and revise.
```
OS: Raspbian GNU/Linux 10 (buster)
OS architecture: 32-bit
Pi-Apps local commit ID: ece04dfc42eeeaa763bb367764bd1c3aa5dba680
Kernel: armv7l 5.10.52-v7l+
Device model: Raspberry Pi 4 Model B Rev 1.1
Raspberry Pi OS image version: 2021-05-07
Language: de_DE.UTF-8

BEGINNING OF LOG FILE:
-----------------------

Installing CommanderPi...
Running pkg-install...
Waiting until APT locks are released... Done
Running sudo apt update...

Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
All packages are up to date.

Creating dummy deb...
Waiting until APT locks are released... Done

Installing dummy deb...

The following additional packages will be installed:
  blt libimagequant0 libtcl8.6 libtk8.6 python3-pil python3-pil.imagetk
  python3-tk tk8.6-blt2.5
Suggested packages:
  blt-demo tcl8.6 tk8.6 python-pil-doc python3-pil-dbg python3-pil.imagetk-dbg
  tix python3-tk-dbg
Recommended packages:
  python3-olefile
The following NEW packages will be installed:
  blt libimagequant0 libtcl8.6 libtk8.6 pi-apps-08f5abf3 python3-pil
  python3-pil.imagetk python3-tk tk8.6-blt2.5
0 upgraded, 9 newly installed, 0 to remove and 0 not upgraded.
(Reading database ... 
(Reading database ... 195186 files and directories currently installed.)

Apt finished.
Klone nach 'CommanderPi' ...
/home/pi/CommanderPi
Checking for tkinter: Status: install ok installed
sudo: pip3: Befehl nicht gefunden
Paketlisten werden gelesen...
Abh\u00e4ngigkeitsbaum wird aufgebaut....
Statusinformationen werden eingelesen....
git ist schon die neueste Version (1:2.20.1-2+deb10u3).
python3-pil ist schon die neueste Version (5.4.1-2+deb10u2).
python3-pil wurde als manuell installiert festgelegt.
python3-pil.imagetk ist schon die neueste Version (5.4.1-2+deb10u2).
python3-pil.imagetk wurde als manuell installiert festgelegt.
python3-tk ist schon die neueste Version (3.7.3-1).
python3-tk wurde als manuell installiert festgelegt.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
/home/pi/CommanderPi
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/pi/CommanderPi/src/start.sh
Icon=/home/pi/CommanderPi/src/icons/Icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /home/pi/Desktop/commanderpi.desktop
Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Desktop/commanderpi.desktop'
chown: Zugriff auf '/home/pi/Desktop/commanderpi.desktop' nicht m\u00f6glich: Datei oder Verzeichnis nicht gefunden
chmod: Zugriff auf '/home/pi/Desktop/commanderpi.desktop' nicht m\u00f6glich: Datei oder Verzeichnis nicht gefunden
failed to run install.sh!

Failed to install CommanderPi with install script!
\u25b2 Need help? Copy the ENTIRE terminal output or take a screenshot.
Please ask on Github: https://github.com/Botspot/pi-apps/issues/new/choose
Or on Discord: https://discord.gg/RXSTvaUvuu
```

Jack477, 2021-08-23:
Looks like desktop file didn't created. It's strange because open(<dir_name>, w) should create file if it not exist already.
`Traceback (most recent call last):
  File "c_desktop.py", line 20, in <module>
    f = open(d_dir, "w+")
FileNotFoundError: [Errno 2] No such file or directory: '/home/pi/Desktop/commanderpi.desktop'`
I've added exception handling for that

Botspot, 2021-08-28:
> Looks like desktop file didn't created. It's strange because open(<dir_name>, w) should create file if it not exist already.
> `Traceback (most recent call last): File "c_desktop.py", line 20, in [...]

Today we received another similar report:
```
OS: Ubuntu 21.04
OS architecture: 64-bit
Pi-Apps local commit ID: f02b85eee9770bf689304b37b2c0d952f864319f
Kernel: aarch64 5.11.0-1016-raspi
Device model: Raspberry Pi 4 Model B Rev 1.4
Language: de_DE.UTF-8

BEGINNING OF LOG FILE:
-----------------------

Installing CommanderPi...
Running pkg-install...
Waiting until APT locks are released... Done
Running sudo apt update...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Hit:1 http://ports.ubuntu.com/ubuntu-ports hirsute InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports hirsute-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports hirsute-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports hirsute-security InRelease
Reading package lists...
Building dependency tree...
Reading state information...
188 packages can be upgraded. Run 'apt list --upgradable' to see them.
Some packages can be upgraded. Please consider running sudo apt full-upgrade.

Creating dummy deb...
Waiting until APT locks are released... Done

Installing dummy deb...

The following packages were automatically installed and are no longer required:
  cryptsetup-bin dctrl-tools dmeventd dmraid dpkg-repack efibootmgr
  gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common grub-efi-arm64
  grub-efi-arm64-bin grub-efi-arm64-signed grub2-common kpartx kpartx-boot
  libdebian-installer4 libdevmapper-event1.02.1 libdmraid1.0.0.rc16
  liblvm2cmd2.03 libtimezonemap-data libtimezonemap1 lvm2 os-prober
  python3-icu python3-pam rdate thin-provisioning-tools
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  blt libtk8.6 libxss1 python3-pil python3-pil.imagetk python3-tk tk8.6-blt2.5
Suggested packages:
  blt-demo tk8.6 python-pil-doc python3-pil-dbg python3-pil.imagetk-dbg tix
  python3-tk-dbg
The following NEW packages will be installed:
  blt libtk8.6 libxss1 pi-apps-08f5abf3 python3-pil.imagetk python3-tk
  tk8.6-blt2.5
The following packages will be upgraded:
  python3-pil
1 upgraded, 7 newly installed, 0 to remove and 187 not upgraded.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 184980 files and directories currently installed.)

Apt finished.
Cloning into 'CommanderPi'...
/home/oskar/CommanderPi
Checking for tkinter: Status: install ok installed
sudo: pip3: command not found
Paketlisten werden gelesen\u2026
Abh\u00e4ngigkeitsbaum wird aufgebaut\u2026
Statusinformationen werden eingelesen\u2026
git ist schon die neueste Version (1:2.30.2-1ubuntu1).
python3-pil ist schon die neueste Version (8.1.2-1ubuntu0.1).
python3-pil wurde als manuell installiert festgelegt.
python3-pil.imagetk ist schon die neueste Version (8.1.2-1ubuntu0.1).
python3-pil.imagetk wurde als manuell installiert festgelegt.
python3-tk ist schon die neueste Version (3.9.5-0ubuntu3~21.04).
python3-tk wurde als manuell installiert festgelegt.
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr ben\u00f6tigt:
  cryptsetup-bin dctrl-tools dmeventd dmraid dpkg-repack efibootmgr
  gir1.2-timezonemap-1.0 gir1.2-xkl-1.0 grub-common grub-efi-arm64
  grub-efi-arm64-bin grub-efi-arm64-signed grub2-common kpartx kpartx-boot
  libdebian-installer4 libdevmapper-event1.02.1 libdmraid1.0.0.rc16
  liblvm2cmd2.03 libtimezonemap-data libtimezonemap1 lvm2 os-prober
  python3-icu python3-pam rdate thin-provisioning-tools
Verwenden Sie \u00bbsudo apt autoremove\u00ab, um sie zu entfernen.
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 187 nicht aktualisiert.
/home/oskar/CommanderPi
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/oskar/CommanderPi/src/start.sh
Icon=/home/oskar/CommanderPi/src/icons/Icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /home/oskar/Desktop/commanderpi.desktop
Couldn't create desktop shortcut!
Save menu shortcut to /usr/share/applications/commanderpi.desktop
chown: cannot access '/home/oskar/Desktop/commanderpi.desktop': No such file or directory
chmod: cannot access '/home/oskar/Desktop/commanderpi.desktop': No such file or directory
failed to run install.sh!

Failed to install CommanderPi with install script!
\u25b2 Need help? Copy the ENTIRE terminal output or take a screenshot.
Please ask on Github: https://github.com/Botspot/pi-apps/issues/new/choose
Or on Discord: https://discord.gg/RXSTvaUvuu
```

Botspot, 2021-08-30:
Here's another person's error log:
```
OS: Raspbian GNU/Linux 10 (buster)
OS architecture: 32-bit
Pi-Apps local commit ID: 127eeab23cdd6acf3b4837319017ee85f71e7135
Kernel: armv7l 5.10.52-v7+
Device model: Raspberry Pi 3 Model B Rev 1.2
Raspberry Pi OS image version: 2021-05-07
Language: es_ES.UTF-8

BEGINNING OF LOG FILE:
-----------------------

Installing CommanderPi...
Running pkg-install...
Waiting until APT locks are released... Done
Running sudo apt update...

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists...
Building dependency tree...
Reading state information...
All packages are up to date.

Creating dummy deb...
Waiting until APT locks are released... Done

Installing dummy deb...

The following NEW packages will be installed:
  pi-apps-08f5abf3
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 95589 files and directories currently installed.)

Apt finished.
Clonando en 'CommanderPi'...
/home/pi/CommanderPi
Checking for tkinter: Status: install ok installed
sudo: pip3: command not found
Leyendo lista de paquetes...
Creando árbol de dependencias...
Leyendo la información de estado...
git ya está en su versión más reciente (1:2.20.1-2+deb10u3).
python3-pil ya está en su versión más reciente (5.4.1-2+deb10u2).
python3-pil.imagetk ya está en su versión más reciente (5.4.1-2+deb10u2).
python3-tk ya está en su versión más reciente (3.7.3-1).
fijado python3-tk como instalado manualmente.
0 actualizados, 0 nuevos se instalarán, 0 para eliminar y 0 no actualizados.
/home/pi/CommanderPi
[Desktop Entry]
Name=CommanderPi
Comment=System info and overclocking
Exec=/home/pi/CommanderPi/src/start.sh
Icon=/home/pi/CommanderPi/src/icons/Icon.png
Categories=Utility;
Version=1.0
Type=Application
Terminal=false
StartupNotify=true
Save desktop shortcut to /home/pi/Desktop/commanderpi.desktop
Couldn't create desktop shortcut!
Save menu shortcut to /usr/share/applications/commanderpi.desktop
chown: no se puede acceder a '/home/pi/Desktop/commanderpi.desktop': No existe el fichero o el directorio
chmod: no se puede acceder a '/home/pi/Desktop/commanderpi.desktop': No existe el fichero o el directorio
failed to run install.sh!

Failed to install CommanderPi with install script!
▲ Need help? Copy the ENTIRE terminal output or take a screenshot.
Please ask on Github: https://github.com/Botspot/pi-apps/issues/new/choose
Or on Discord: https://discord.gg/RXSTvaUvuu
```

Botspot, 2021-09-01:
Another instance of this issue: https://github.com/Botspot/pi-apps/issues/923#issuecomment-910541512
@jerry8538 seems to indicate that this failure was caused because **the ~/Desktop folder does not exist.**