Jack477/CommanderPi #72: CommanderPi won't start

TogaParty, 2023-01-13:
I tried installing from your install instructions here, as well as installing from Pi Apps (this is the current install).  I get an hourglass, then nothing.  Per # 39, I have included the output from the requested command.

Looking below, I see _tkinter.TclError: couldn't connect to display ":10.0".  I am logged in over a remote session, this is an RPi 4 8GB running 64 bit, I do not have a monitor connected.

pi@FoundryVTTPPE:~ $ sudo python3 ${HOME}/CommanderPi/src/main.py ${HOME}
Here is home_path /home/pi
Here is app-1 Version 1.0
Exist and read
Country code is:
country US: DFS-FCC
Here is ad: pieeprom-2022-03-10.bin
Apr 29 2021
/home/pi/CommanderPi/src
File is exist!
path is /home/pi/CommanderPi/src
No protocol specified
No protocol specified
Traceback (most recent call last):
  File "/home/pi/CommanderPi/src/main.py", line 26, in <module>
    main()
  File "/home/pi/CommanderPi/src/main.py", line 22, in main
    start = g.Window()
  File "/home/pi/CommanderPi/src/gui.py", line 673, in __init__
    master = tk.Tk()
  File "/usr/lib/python3.7/tkinter/__init__.py", line 2023, in __init__
    self.tk = _tkinter.create(screenName, baseName, className, interactive, wantobjects, useTk, sync, use)
_tkinter.TclError: couldn't connect to display ":10.0"

TogaParty, 2023-01-13:
The following command fixes this particular issue.  If you can include it in your scripts, it might save some users some time, but I'm a linux noob, so I don't know what risk that might pose.

sudo cp ~/.Xauthority ~root/

miguelangelramirez, 2024-08-03:
The above line of code works, it does revert for some reason once you reboot. I fixed it by adding that line  to **src/start.sh** before running **src/main.py**.